CanyoneroRobotics/TFM #1: Raspberry Pi 3 Ubuntu installation and basic configuration

simium, 2016-05-19:
Follow: https://wiki.ubuntu.com/ARM/RaspberryPi

simium, 2016-05-23:
The Ubuntu image available for the Pi3 is not official and there's a lack of support for some functionality (wifi configuration, partition resize to use the whole SD card... )

I'm thinking about switching to Raspbian, which is official, provided by the Raspberry people, and can run ROS Indigo: http://www.jonathanfritz.ca/2015/10/24/installing-ros-on-a-raspberry-pi/

Are we ok with using ROS Indigo? Do you prefer Jade? There's a short list of differences: http://wiki.ros.org/Distributions

marcgenevat, 2016-05-23:
As we talked on Thursday with German, Indigo is the most stable version and it will have support until 2019, so I think it's our best option for educational purpose.

So, for me, go for it! ;)

oriolorra, 2016-05-23:
If Raspbian image is official and supports ROS Indigo (as German said), it is fine.

I agree!

simium, 2016-05-23:
Let's start with Indigo, we can move to Jade later if @oriolorra thinks we need a more recent version of Gazebo.

oriolorra, 2016-05-23:
I think that we could manage it with the actual version of Gazebo. A lot of simulator are implemented with it, like Husky of Clearpath Robotics.

simium, 2016-05-23:
Agree!

marcgenevat, 2016-05-23:
Let's do it then! Look forward to see if you succeed with the installation. Let us know if you have problems, but you should be fine.

simium, 2016-05-23:
We will use Ubuntu in the future on-board computer (the Intel NUC), but for now we are going to use Raspbian: https://www.raspbian.org/

closed 2016-07-25 by simium