CanyoneroRobotics/TFM #3: Arduino Mega 2560 ROSSerial installation

simium, 2016-05-19:
See: http://wiki.ros.org/rosserial_arduino/Tutorials

marcgenevat, 2016-05-19:
És el que vam fer amb l'Andreu no? O té alguna diferència?

simium, 2016-05-19:
En principi és el mateix però s'ha de tornar a integrar tot! Per exemple, amb l'Arduino Uno havíem de fer reset manual abans de llençar ROS.

Hem d'anar trobant aquests detalls...

marcgenevat, 2016-05-19:
La majoria de components del tutorial els tenim a Make&Learn, així que si voleu el podem fer i deixar-ho al repo com a introducció a Arduino amb ROS. Potser queda bé de cara a la memòria.

Se'm acudeix que podríem posar algun sensor i actuador bàsic al Canyonero precisament per iniciar-se. No és lu mateix fer un publisher o un listener simplement amb missatges string que poder encendre leds o moure servos o rebre dades d'algun sensor bàsic rollo ultrasons, temperatura, humitat, llum,...

simium, 2016-05-19:
Em sembla perfecte! Proporciona a l'Oriol aquests sensors ASAP! :P

marcgenevat, 2016-05-19:
Quins sensors i actuadors posaríeu doncs?

Molaria tenir dades de:

- IMU
- GPS
- Visió (Kinect/RealSense)
- Odometria (Encoders)
- Ultrasons?
- ...

Pel que fa actuadors:

- Servo per donar-li YAW al sensor d'ultrasons?
- Tira de leds rollo neones? XD
- LEDs per indicar el nivell de bateria?
- Pantalla LCD?
- ...

Ja direu què us molaria i mirem d'aconseguir-ho, a través de la meva empresa o per on sigui.

oriolorra, 2016-05-23:
Done!

motor controlled with Joy and get encoder ->( commit 60addb70302a5986b7fde6990bfda8de4252f9d3)