CanyoneroRobotics/TFM #4: Arduino Mega 2560 + Motor shield integration

simium, 2016-05-19:
See: https://learn.adafruit.com/adafruit-motor-shield-v2-for-arduino/using-dc-motors

marcgenevat, 2016-05-19:
Us passo el codi que vaig utilitzar en el projecte de la plataforma omnidireccional de tres rodes que controlavem amb LeapMotion la qual tenia un parell de les shield de Adafruit. Espero que et serveixi @oriolorra!

`#include <Wire.h>
#include <Adafruit_MotorShield.h>
#include "utility/Adafruit_PWMServoDriver.h"

Adafruit_MotorShield AFMS = Adafruit_MotorShield(); 

Adafruit_DCMotor *myMotor1 = AFMS.getMotor(1);
Adafruit_DCMotor *myMotor2 = AFMS.getMotor(2);
Adafruit_DCMotor *myMotor3 = AFMS.getMotor(3);

char movement;
int vel = 255;

void setup() {
  Serial.begin(9600);           // set up Serial library at 9600 bps
  Serial.println("Adafruit Motorshield v2 - DC Motor test!");

  AFMS.begin();  // create with the default frequency 1.6KHz
  //AFMS.begin(1000);  // OR with a different frequency, say 1KHz
}

void Forward() {
  myMotor1->setSpeed(vel);
  myMotor2->setSpeed(vel);
  
  myMotor1->run(FORWARD);
  myMotor2->run(BACKWARD);
  myMotor3->run(RELEASE);
  }

void Backward() {
  myMotor1->setSpeed(vel);
  myMotor2->setSpeed(vel);
  
  myMotor1->run(BACKWARD);
  myMotor2->run(FORWARD);
  myMotor3->run(RELEASE);
  }

void Left() {
  myMotor1->setSpeed(vel);
  myMotor2->setSpeed(vel);
  myMotor3->setSpeed(vel);
  
  myMotor1->run(FORWARD);
  myMotor2->run(FORWARD);
  myMotor3->run(BACKWARD);
  }
  
void Right() {
  myMotor1->setSpeed(vel);
  myMotor2->setSpeed(vel);
  myMotor3->setSpeed(vel);
  
  myMotor1->run(BACKWARD);
  myMotor2->run(BACKWARD);
  myMotor3->run(FORWARD);
  }
  
void Left_Rotation() {
  myMotor1->setSpeed(vel);
  myMotor2->setSpeed(vel);
  myMotor3->setSpeed(vel);
  
  myMotor1->run(FORWARD);
  myMotor2->run(FORWARD);
  myMotor3->run(FORWARD);
  }
  
void Right_Rotation() {
  myMotor1->setSpeed(vel);
  myMotor2->setSpeed(vel);
  myMotor3->setSpeed(vel);
  
  myMotor1->run(BACKWARD);
  myMotor2->run(BACKWARD);
  myMotor3->run(BACKWARD);
  }
 
void Break() {
  myMotor1->run(RELEASE);
  myMotor2->run(RELEASE);
  myMotor3->run(RELEASE);
  } 

void loop() {
  uint8_t i;
  while(!Serial.available());
  movement = Serial.read();
  switch (movement) {
    case 'B':
      Backward();
      while(movement == 'B') {movement = Serial.read();}
      Break();
      break;
    case 'F':
      Forward();
      while(movement == 'F') {movement = Serial.read();}
      Break();
      break;
    case 'L':
      Left();
      while(movement == 'L') {movement = Serial.read();}
      Break();
      break;
    case 'R':
      Right();
      while(movement == 'R') {movement = Serial.read();}
      Break();
      break;
    case 'X':
      Left_Rotation();
      while(movement == 'X') {movement = Serial.read();}
      Break();
      break;
    case 'Y':
      Right_Rotation();
      while(movement == 'Y') {movement = Serial.read();}
      Break();
      break; 
    default:
      Break();
      break;
      }
}`

oriolorra, 2016-05-20:
See: [http://www.dfrobot.com/wiki/index.php/Encoder_Adapter_FIT0324](url)

oriolorra, 2016-05-23:
Done!

motor controlled with Joy and get encoder ->( commit 60addb70302a5986b7fde6990bfda8de4252f9d3)