CanyoneroRobotics/TFM #8: Components distribution

marcgenevat, 2016-05-23:
We'll have the hardware distributed on the first and second floor and the top (lidar).

First suggestion:

Raspberry PI 3 and IMU should be together and at the center of the robot. The same for the batteries in order to have the gravity center centralized.

Then, the Arduino+shield should be at the second floor, along with the router and the 3D camera, currently the kinect, which it'll be at the frontal side with a gap at the methacrilate plate for the different cameras.

The encoder boards can be at the second floor but below, so don't occupy too much space and the wires can go through a gap on the second floor between the Raspberry PI and the Arduino, where a lot of wires will be able to go through, like the power supply.

The lidar could be at the very frontal top of the robot having the possibility to scan around using a 270º or 360º.

marcgenevat, 2016-05-23:
Maybe we should reconsider the position of the router, so we don't have a lot of space at the second floor, less if we use the kinect. It would be much better using the Real Sense camera, it's much smaller.

simium, 2016-05-23:
As we commented offline, we could set the router as if it was a backpack on the robot's back, that way we could improve the wifi coverage. There are wall mounts available for our WRT54G router, or we could design and print our own :sunglasses: https://www.google.es/search?q=linksys+wrt54g+wall+mount&source=lnms&tbm=isch

marcgenevat, 2016-05-23:
But then the router will be out of the robot, right? With no protection from the outside... Isn't that a problem?

simium, 2016-05-23:
Oh, that's true! For the outdoors applications it would be too dangerous... maybe we could get a smaller router later in the project if free space is a serious issue.

marcgenevat, 2016-05-23:
Yeah... maybe we can find a smaller router that works as good as this one.