CanyoneroRobotics/TFM #21: Motion system

marcgenevat, 2016-07-23:
The motion system has been causing some troubles from the beginning.

[image]

Due to the size of the wheel, an adaptador for the saft had to be made. The motor we're using has a very short and small shaft (ø6mm x 14mm) so we had to couple a larger shaft in order to hold and resist the weight of the robot, which it may be pretty heavy.

The first idea was to attach a ø8mm shaft to the motor's shaft with a flexible or rigid coupler. Then, it would have a plastic bearing holding the larger shaft and it would attach to the wheel using the adapter piece created and a grub screw from the inside part of the wheel.

[image]

After a meeting with my tutor Carlos Rosales, we talked about the importance of accessibility and ease to mount and unmount parts. Also, with Jordi Serra, UVic workshop's chief, talked about simplifying some parts and increasing the resistance of the system with a larger shaft for the wheel.

So, after several options and ideas which its dismissed is going to be explained at the final report, I have decided to design and implement this system:

[image]
[image]

The wheel's shaft is now ø20mm with a socket of ø18mm for the wheel's inside part. As it's that huge, the shaft itself can act as a coupler for the motor's shaft. The wheel's shaft will be attached to the wheel using a single adapter piece from the wheel's outside part with a passing grub screw. There will be an igus ball bearing will be at the middle of the wheel's shaft onto the aluminum profile.

The new wheel's shaft adapter for the outside (has to avoid the tire's air nozzle) looks like that:

[image]

The whole assembly has this appearance right now:

[image]

If my tutor Carlos validate this new design, I'm going to build it this next week and start redesigning the plates and covers to improve their accessibility and ease to mount and unmount.

carlosjoserg, 2016-07-23:
LGTM! :+1:

Left a couple of comments on the F360 project.

marcgenevat, 2016-08-03:
last week i had enough time to make all the needed parts to build the motion system.

so now we have the 4 motor plates, the 4 turned shafts and 2 wheel-shaft adapter, i'll have to print two more by myself.

[image]

in a few days we should receive the igus bearings. then we'll be able to mount the motion system to the chassis.